Jan-IngenHousz-Institute/open-jii #119: As a PS researcher I want to add a multispeq sensor to an experiment

Chris-INFONL, 2025-03-20:
## User Story

**WHO:** PS researcher

**WHAT:** Add a multispeq sensor to an experiment, in order to start taking measurements with it

**WHY:** In order to start measuring with the multispeq sensor, it first needs to be added to an experiment. This should lead to (next user story!) enabling communication with the sensor device, which relies on setting it up with a relevant protocol.

## Acceptance Criteria

- [Clearly define the conditions that must be met for the story to be considered complete]
- [Include primary user flow and relevant alternative flows]
- [Explicitly state critical test cases and business logic rules, if applicable]

## Dependencies & Risks

- [Identify dependencies with other features, teams, or external resources]
- [List potential risks or issues and how they might be mitigated]

## Additional Context

- [Provide any additional relevant information, attachments, or references]

dominikvrbic, 2025-03-21:
i do not think this is needed since you are not adding multispeq to an experiment you are doing a mesurment with multispeq for an experiment, diffrent from ambit

Chris-INFONL, 2025-03-24:
I've added this specific user story, which I think should be easy, since most every project should include the multispeq sensor as an option to take measurements with. If we assume that the multispeq is 'pre-installed' for every user, that will make this user story super easy. Namely, the user should then only have to select the multispeq from a list and _voila_: it is added and ready to start measurements. In other words, this is _a step in the user flow_ for setting up an experiment, but likely a very easy step.

dominikvrbic, 2025-03-24:
Your phone either has a multispeq plugged into it at the moment, or it doesn't

Chris-INFONL, 2025-05-07:
I've updated screens to clarify some things and removed the elements that are not necessary for now (towards MVP):

Screen 1: "new experiment" button is removed. User can only open an existing experiment.

Screen 2: Changed the toggle to "show all public experiments" - clarifying that this list all public experiments (including those the user is not owner or member of)
Also changed the column 'owner' to "owner/member" to show the distinction between those. Members can only take measurements and see limited (or no) past measurement data.

Screen 3: Removed the 'data feed' part, which is not necessary for now. (Will be applicable later, when the Ambit sensor is added, post-MVP)
To clarify: the button "view measurement data" is now updated to "view past measurement data". If logged in as an owner, you can see all past measurements that the experiment contains. If logged in as a member, presumably you can only see past measurements on your own device, or none at all (only able to take new measurements). Up for debate.

Screen 4: Devices need to be registered first. This example shows 2 earlier registered device, one which is outdated, another which is currently out of bluetooth range. Hence, the user needs to register a new one (the one currently held by the user)

Screen 5 shows this process of registering a new device. As an example, it shows other sensors transmitting via bluetooth.

Screen 6: The option to 'edit protocol' has been removed. There was an idea to select another protocol from a dropdown list, but that will make things too complicated. The user can definitely not edit the code of the protocol from their smartphone.