Jan-IngenHousz-Institute/open-jii #180: Connect MultispeQ 2.0 to the iOS version of openJII Mobile App

Chris-INFONL, 2025-04-24:
## User Story

**WHO:** openJII mobile app user with MultispeQ 2.0

**WHAT:** Connect the MultispeQ 2.0 (preferably via bluetooth) to an iOS device

**WHY:** Some users in the field are expected to have only an iOS device (e.g. iPhone 12 and newer) at their disposal when trying to take measurements in the new openJII mobile app, using the MultispeQ 2.0 sensor device. This device currently does not connect to iPhones due to Bluetooth compatibility issues. There might be a solution in adding a peripheral devices to overcome these compatibility problems.

## Acceptance Criteria

- MultispeQ 2.0 can be registered and connected to the new mobile app on iOS
- MultispeQ 2.0 show correct connectivity on the iOS version of the mobile app
- [Include primary user flow and relevant alternative flows]
- [Explicitly state critical test cases and business logic rules, if applicable]

## Dependencies & Risks

- [Identify dependencies with other features, teams, or external resources]
- [List potential risks or issues and how they might be mitigated]

## Additional Context

- [Provide any additional relevant information, attachments, or references]

Chris-INFONL, 2025-07-07:
Not sure anymore if this is covered in any of the other user stories about connecting multispeq to a phone - in this case specifically iOS

dominikvrbic, 2026-01-06:
wont do