Jan-IngenHousz-Institute/open-jii #322: Saving a new experiment gives error when name is not unique

DorienvdBerg, 2025-06-05:
**To Reproduce**
Create an experiment with a certain name, save. 
Create another experiment with the same name
Check what happens

**Expected behavior**
I expect I get a clear error message already when I fill in the same name in the field. Saying I cannot create another experiment with a name that already exists. 
When I call the API in Postman, I see a response that the name must be unique. 

**Actual**
I see an unclear error, but I have the suspicion that this is about creating a new experiment with an already existing name. 

[image]

dominikvrbic, 2025-06-05:
The unclear error is from the dev mod, the rest I agree tho the customer error msg is for later will also be in the cms, for right now it should just show what ever the backend is returning

DorienvdBerg, 2025-06-11:
Tested, gives an error message now (inline).
After saving.