Jan-IngenHousz-Institute/open-jii #501: Create new organization

Chris-INFONL, 2025-07-31:
## User Story

**WHO:** Experiment owner

**WHAT:** Can create a new organization

**WHY:** So field managers can have their teams of measurers recognized as 'organizations'. This makes it easy to attach the organization to the experiment, which enables the team of measurers to be linked to an experiment, giving them all the permission needed to take measurements. Multiple organizations can be linked to a single experiment (e.g. multiple measurement teams)

## Acceptance Criteria

- Experiment owner can attach organizations (from a list) to an experiment, in the 'settings' tab. The 'organization' segment is located next to the 'members' segment.
- - If the organization is not yet registered, the experiment owner can add it in a dropdown list
- - For adding an organization, a separate modal appears
- In the 'add new organization' modal, the text fields  (max 255 char) 'organization name' (mandatory) and 'affinity to institute' (optional), and 'location' (optional) are visible
- - 'Affinity to institute' is an optional text/search field. It allows the user to type and search for institutes (such as universities) from a database **(external?)**, which automatically appear as the user starts typing. 
- - 'location' is an optional text/search field. It allows the user to type and search for locations (cities, villages) world wide, which automatically appear as the user starts typing
- Below the 'organization name', 'affinity to institute' and 'location', is an area in which members can be added to an organization. This is done with a search-text field, which lists known openJII as the user starts typing. Adding an openJII user adds it to a list.
- - The 'organization member list' can contain multiple registered openJII users. The first to be added is 'organization leader' by default. This person will become the 'owner' of the organization. The label 'leader' can be changed to 'member'. However, at least one user in the organization has to be the leader.
- After adding, editing and removing organization members (and a leader), the organization can be saved by clicking 'save' button in the modal. The button 'cancel' takes the user back to the 'settings' tab of the experiment.
- If a new organization was added (saved), it will appear in the dropdown list as an organization. Clicking this adds it to the experiment.
- The 'leader' of an organization is notified of being the leader. This person then has administrative rights over their organization (group of users). The organization can be edited only by the leader.

## Dependencies & Risks

- Public list of knowledge and research institutes (universities, etc) needs to be available for 'affinity to institute' option
- Setting up organizations adds a new layer of complexity and user account management

## Additional Context

- [Provide any additional relevant information, attachments, or references]

Chris-INFONL, 2025-09-30:
Also: think about the distinction between 'organizations' (eg Wageningen University) and 'teams' (eg a measuring team of 12 people in a specific location, contributing to multiple experiments)

dominikvrbic, 2026-10-20:
Closing: canceled in Linear as [OJD-317](https://linear.app/openjii/issue/OJD-317) on 26 August 2026.

Canceled with its parent epic OJD-316 (#500). Organizations are now covered by the access and organization work in OJD-1638.

The openJII team tracks its work in Linear, so any follow-up happens there.

closed 2026-10-20 by dominikvrbic